simprecicchiani/ThinkPad-T460s-macOS-OpenCore #103: Hiberination 25 cmos solution (work for my t460s)

mhl221135, 2021-04-03:
Sup guys! I got working hibernation on my T490 with Whiskey Lake cpu using OpenCore bootloader. As I saw "RTC Lock" chipset option isn't the main cause of checksum error, due to I also had this behaviour on my board with disabled RTC Lock in PCH settings. Higher bank (128-255) used by Lenovo for sensitive or diagnostic information like setup settings checksum, that's why checksum error thrown, if any data had been written into it.
So first of all you should use RTCMemoryFixup and HibernationFixup.
Then remove ranges from rtc-blacklist, if you use any, and add 80-AB range into exclude for RTCMemoryFixup (rtcfx_exclude=80-AB). This actions should fix checksum error. After this you need to set HibernateMode to NVRAM.
The main problem - black screen after wake could be fixed by enabling CSM support inside BIOS Setup Utilty, seems it affects memory map and do something tricky which resolves restore from hibernation file into black screen.
Also you can check my set of quirks
[config.plist](https://ghostbin.co/paste/q4h4a)

_Originally posted by @savvamitrofanov in https://github.com/tylernguyen/x1c6-hackintosh/issues/44#issuecomment-696689768_

mhl221135, 2021-04-03:
Also i have no issues with touchpad after sleep and intel wifi also working fine

closed 2021-04-06 by mhl221135

vihor3, 2024-01-16:
Thank you very much. The source you quoted is my model. I have encountered this problem and am seeking a solution