simprecicchiani/ThinkPad-T460s-macOS-OpenCore #107: OpenCore old version problem

iTer99, 2021-04-18:
i dont know, but for some reasion OpenCore Configurator say "You seem to have an old version of OpenCore bootloader. This app creates a configuration file for 0.6.8 or higher versions". and i found command in Terminal for check current OpenCore version `nvram 4D1FDA02-38C7-4A6A-9CC6-4BCCA8B30102:opencore-version`. 
4D1FDA02-38C7-4A6A-9CC6-4BCCA8B30102:opencore-version	REL-066-2021-02-02

duszmox, 2021-04-18:
It seems, i forgot to copy the NVRAM version over to the 0.6.8 config. I will fix it in the next release soon.

iTer99, 2021-04-18:
> Có vẻ như, tôi đã quên sao chép phiên bản NVRAM sang cấu hình 0.6.8. Tôi sẽ sửa chữa nó trong bản phát hành tiếp theo.

ok thank you so much

closed 2021-04-18 by iTer99