simprecicchiani/ThinkPad-T460s-macOS-OpenCore #110: Keyboard is writing symbols when pressing the Windows key + another key

damaini, 2021-05-20:
Hello

I'm having a weird behavior on my keyboard.  When I press the Windows key + another key.
My understanding is that the Windows key should become the Command key for Apple.
But when I press the Windows key + another key I get symbols.
Here are some examples:
Windows key + z      Ω
Windows key + a      å
Windows key + s      ß
Windows key + c      ç
Windows key + x      ≈
Windows key + v     √

It can be funny, even though is currently annoying the heck our of me.  I had to find a work around and make the left Ctrl in the keyboard become the Command key.

Has anyone experienced this issue?

BTW.
This is a Lenovo Thinkpad T460s  Type 20F9-003CUS.
Everything else is working fine.

duszmox, 2021-05-20:
Yes, this can be confusing. By default a keyboard's windows key is associated to the command key in macOS, but in the EFI, we've made a change to the laptop's built-in keyboard to replicate the original apple layout which means, we swapped the alt keys to the command keys and vise-versa, and now the command is sitting right next to the spacebar, similar to a real MacBook. So your command key is your alt key and your an alt key (a.k.a. the option key in macOS) is your windows key. If you want to cmd+c something you have to use "alt"+c. Also if you want to change it back you can do it easily in System Preferences/Keyboard/Modifier keys [via changing the option to command and the command to option](http://duszmo.f.fazekas.hu/keyboard.png). Hope this wasn't too confusing

duszmox, 2021-05-20:
Also, I'm now starting to wonder, how many people had the same "problem" as you. Maybe I will include a mention of this in the guide in the near future, to prevent this confusion.

damaini, 2021-05-21:
I don't have that keyboard.  How do I install it?  I have just keyboard, no Karabiner.  Did I miss something?

duszmox, 2021-05-21:
> I don't have that keyboard.  How do I install it?  I have just keyboard, no Karabiner.  Did I miss something?

Sorry about that, you should see 'Keyboard' there. I'm using [Karabiner](https://karabiner-elements.pqrs.org) to remap some other keys, for example my insert key to the eject key. But Karabiner aside, you should change the settings of the 'Keyboard'

closed 2021-08-05 by duszmox