simprecicchiani/ThinkPad-T460s-macOS-OpenCore #112: Bootcamp and Startup Disk in MacOS

iTer99, 2021-05-23:
I was check this, Bootcamp in Windows work but in MacOS doesn't show up
https://dortania.github.io/OpenCore-Post-Install/multiboot/bootcamp.html
[image]

duszmox, 2021-05-25:
You can set, the default boot OS in the OpenCore bootloader with CTRL+Enter. [Here is some more info](https://www.reddit.com/r/hackintosh/comments/ep0qag/opencore_054_now_supports_default_boot_option/?utm_source=share&utm_medium=ios_app&utm_name=iossmf)

iTer99, 2021-07-06:
[image]

1) Use mini Windows via USB Boot have CMD:
diskpart
list disk
select disk 0 (0 is Hard Disk use have Hackintosh)
create part primary size=450
format quick fs=ntfs label=Recovery (creat partition Recovery for Windows)
assign letter=R
set id=“de94bba4-06d1-4d40-a16a-bfd50179d6ac”
gpt attributes=0x8000000000000001
create part msr size=16 (partition MSR, I don't know what is it, but it need for Windows partition)
create part primary (partition for install Windows)
format quick fs=ntfs label=Windows
assign letter=M
2) Mount any ISO Windows then open Setup.exe and install Windows normal

closed 2021-07-07 by iTer99