simprecicchiani/ThinkPad-T460s-macOS-OpenCore #116: Apple progress bar after login password

iTer99, 2021-06-03:
When i turn on device, my device request password and then show apple progress bar. How can i change it, i want when i turn on device it run apple progress bar first and then login password? thank you so much

mhl221135, 2021-06-03:
You using hibernate mode ?

iTer99, 2021-06-03:
> You using hibernate mode ?

i don't know, i shutdown and turn on

mhl221135, 2021-06-03:
> > You using hibernate mode ?
> 
> i don't know, i shutdown and turn on
Strange, maybe its becouse of slow ssd, dont know

iTer99, 2021-06-03:
> > > You using hibernate mode ?
> > 
> > 
> > i don't know, i shutdown and turn on
> > Strange, maybe its becouse of slow ssd, dont know

you can see in this video, when him turn on laptop the apple logo will show with progess bar and then password request login, but on my laptop, i turn it on the password request first and then apple logo show with progress bar
https://www.youtube.com/watch?v=SnuQjuIrfc0

duszmox, 2021-06-03:
You are using FireVault disk encryption, if you turn it off, it will act normally

iTer99, 2021-06-03:
> You are using FireVault disk encryption, if you turn it off, it will act normally

exactly, thank you so much

closed 2021-06-03 by iTer99