simprecicchiani/ThinkPad-T460s-macOS-OpenCore #119: Screen touch

iTer99, 2021-06-24:
I found VooDooI2C.kext support enable touch screen, how can I add this kext to my device. because my T460s have touch screen working.
https://github.com/VoodooI2C/VoodooI2C

mhl221135, 2021-06-24:
I tried this kext, unfortunately it is not working with our screen(

askwakhid, 2021-08-04:
> I tried this kext, unfortunately it is not working with our screen(

Dear Mikhail,

First, I would say Thank you for you and this EFI creator & maintener.
Im sorry for my bad english, but maybe I have some informations.

I Have T460s touchscreen.
I found the touchscreen work in Catalina with clover.
The EFI creator is Simone too. : https://github.com/markgotit/Thinkpad-T460s-MacOS-Catalina-Clover

But, the issue is touchscreen only works on catalina intaller GUI.
I tried process to catalina installation until finish, after installation finish, the touchscreen didnt works.

Because of that, I trust the EFI can activate the touchsreen but maybe need USB re-maping.
After searcing information in google, I found touchsreen driver apps called http://touch-base.com (paid subscriptions).

I tried touch base apps and the touchsreen works perfecly. 

This is my demo video with Catalina : https://video.fcgk6-2.fna.fbcdn.net/v/t42.1790-2/228158913_130343669260892_2298833873472240995_n.mp4?_nc_cat=110&ccb=1-3&_nc_sid=985c63&efg=eyJ2ZW5jb2RlX3RhZyI6InN2ZV9zZCJ9&_nc_eui2=AeHxZ9ObjPrUwMUjF9yN5xbT0bdTLc5SR5nRt1MtzlJHmYNneE9IyRECmHVCLzDxaMHdp8967ZTO5--CtDd5PCV0&_nc_ohc=FU8mhJhtH0IAX-cZPIC&_nc_rml=0&_nc_ht=video.fcgk6-2.fna&oh=a4035916a2d89cdbcea7bb3414f57285&oe=610A321E

I think the touchscreen is supported by Catalina but need some tweak or configurations. 

Then, I tried to install Big sure and touch-base.com with hope the touchsreen can works. But I wrong and the touchsreen didnt works.
Because of my knowledge limitation in Hackintosh, I hired someone to check my Hackintosh.

He still create EFI configuration for my laptop on Big Sur.

Now, My Laptop touchscreen can work with Open Core 0.6.2 and the touch-base.com drivers.
This is my demo video on Big Sur : https://video.fcgk6-2.fna.fbcdn.net/v/t42.1790-2/232046005_902213703736110_386496962506707631_n.mp4?_nc_cat=110&ccb=1-3&_nc_sid=985c63&efg=eyJ2ZW5jb2RlX3RhZyI6InN2ZV9zZCJ9&_nc_eui2=AeHHEG2BqzDmfdvgGv8qw3Lq2a9Jm5xIGfnZr0mbnEgZ-YDSbsycnE7vHEsh7x_rC9yqXBm4vySn365q1DRDIb9q&_nc_ohc=2wEpeEpWUksAX9_Aw-n&_nc_rml=0&_nc_ht=video.fcgk6-2.fna&oh=999300798251f8deebad1744a35504ca&oe=610A4067

I just hope someone can fix the touchsreen on Big Sur, because the touch-base.com driver is so expensive.

I will update after all finished.

Thank you.

askwakhid, 2021-08-04:
> I tried this kext, unfortunately it is not working with our screen(

Dear Mikhail,

First, I would say Thank you for you and this EFI creator & maintener.
Im sorry for my bad english, but maybe I have some informations.

I use your T460s EFI + http://touch-base.com touchscreen driver.

First, we need inject all usb using rehabman usbinjectall.kext.
After reboot, using Hackintool, we will see the touchscreen interface detected as usb HXCI LGD AIT Touch Controller.

Then, Install http://touch-base.com touchscreen driver. 
The touchsreen will works fine. I already test the touch and multi touch gesture works fine.

This is my demo video : https://youtu.be/Ss-QM4aQKAE

But, http://touch-base.com touchscreen driver is not free, we must paid for subscription.

Maybe with this small informations can help the EFI mainterner for making some solutions.

Thank you.

mhl221135, 2021-08-20:
> > I tried this kext, unfortunately it is not working with our screen(
> 
> Dear Mikhail,
> 
> First, I would say Thank you for you and this EFI creator & maintener.
> Im sorry for my bad english, but maybe I have some informations.
> 
> I use your T460s EFI + http://touch-base.com touchscreen driver.
> 
> 1. we need inject all usb using rehabman usbinjectall.kext.
> 2. download this kext and save to EFI/OC/KEXT and dont forget to activate in Conf.plist
>    https://drive.google.com/file/d/13RFk0ngwz31DPD_t2SOChENi09fhCr5Y/view?usp=sharing
> 3. After reboot, using Hackintool, we will see the touchscreen interface detected as usb HXCI LGD AIT Touch Controller.
> 4. Install http://touch-base.com touchscreen driver.
> 5. Reboot
> 6. The touchsreen will works fine. I already test the touch and multi touch gesture works fine.
> 
> This is my demo video : https://youtu.be/Ss-QM4aQKAE
> 
> But, http://touch-base.com touchscreen driver is not free, we must paid for subscription.
> 
> Maybe with this small informations can help the EFI mainterner for making some solutions.
> 
> Thank you.

Thank you for your info, i have same touch and make same steps as you describe but its not working for bigsur( also with touch injected in devices voodooi2chid also not working(

mhl221135, 2021-08-20:
yahoo!!! now its working with voodooi2c, i make manual soon, but only one thing i dont know how to inject its without usbinjectall.kext

mhl221135, 2021-08-20:
> I found VooDooI2C.kext support enable touch screen, how can I add this kext to my device. because my T460s have touch screen working.
> https://github.com/VoodooI2C/VoodooI2C

Its working now, check readme

closed 2021-08-20 by mhl221135