simprecicchiani/ThinkPad-T460s-macOS-OpenCore #125: trackpoint and touchpad lose response after a random time

jackhu9527, 2021-09-03:
Thanks for sharing your achievement, and I was able to use the same EFI folder to run macOS on my T460s without any major issue. Just found that my trackpoint and touchpad will somehow stop working after booting into the system for a while, the trackpoint doesn't respond at all, same goes for the touchpad;While at the same time, I will still be able to use my exteral mouse to do everthing. The internal keyword is working all the time though. I didn't change anything in your EFI folder, except that I've added my own serial number,etc. to make iCloud working.
Feel free to let me know if you need any extra information to look into this, thanks and have a great day!

mhl221135, 2021-09-03:
> Thanks for sharing your achievement, and I was able to use the same EFI folder to run macOS on my T460s without any major issue. Just found that my trackpoint and touchpad will somehow stop working [...]

You using last release of efi?

duszmox, 2021-09-03:
Thanks, for sharing the info. Does this happen after the computer went to sleep? Also next time it does that, can you run this command in the terminal, and send me the results? `grep Voodoo`

jackhu9527, 2021-09-03:
Yes I'm using the latest release out there.

jackhu9527, 2021-09-03:
> Does this happen after the computer went to sleep?

No this happens everytime, either it's cold boot or restart.

> Also next time it does that, can you run this command in the terminal, and send me the results? `grep Voodoo`

So I have run this command in the terminal, but nothing was shown in the terminal then. Please advise, thanks.

jackhu9527, 2021-09-04:
Hi, so yesterday night I re-downloaded the latest release, added my own serial number,UUID, etc. and then replaced the existing OC folder in my EFI partition. And it seems the trackpoint and touchpad have been working normally ever since then. Well now I found that the touchscreen can't be enabled after I added the content of #touchscreen.plist to the config.plist. But I don't really use touchscreen often and maybe this is because I'm running a beta 6 version of macOS Monterey - I can live with that. Thanks for all your kind feedback&support.

closed 2021-09-04 by jackhu9527

cyberfunk, 2022-04-20:
If you have this issue happen again please try disabling the red track-point button in your SMBIOS. I found that this prevents the cursor from "freezing." I think in the past there was a conflict between track-point and touchpad with the kexts that are included.