simprecicchiani/ThinkPad-T460s-macOS-OpenCore #136: Trackpad only working 1 hour then stopping

robertojguerra, 2021-12-01:
Hello... I don't know whether I should post my problem here, in VoodooPS2, VoodooRMI or VoodooI2C, since this project uses all 3 kexts.
I use a Yoga 260... is everyone else on T460 not suffering this problem? After 1 hour of use, the trackpad stops working and I can only continue with a USB mouse.

simprecicchiani, 2021-12-01:
Does this happen with older releases?

robertojguerra, 2021-12-02:
Oh sorry I forgot to add: I am using Monterey and I haven’t tried BigSur.

robertojguerra, 2021-12-22:
I found from the kreizlie GitHub project, that disabling the red trackpoint in BIOS, prevents the trackpad from stopping.

cyberfunk, 2022-04-29:
I have a new issue where the trackpad will stop responding after an undetermined amount of time. My computer does not freeze it just seems like the trackpad disconnects from the system and will not resume functioning until I reboot. I can plug in a USB mouse and continue working when it happens but the trackpad just stops working. Does anyone have a kext or other fix for this behavior?

robertojguerra, 2022-04-29:
Did you try disabling the red “nipple” in the bios?

cyberfunk, 2022-04-29:
Yes I'm the one who came up with that solution to stop the computer from freezing. This is different though. The computer does not freeze but the trackpad simply stops functioning altogether.