simprecicchiani/ThinkPad-T460s-macOS-OpenCore #145: Solve "OCS: Missing key UIScale, context <Output>! Error in Config.plist?

cyberfunk, 2022-02-07:
Can anyone tell me how to get rid of this error in OpenCore Validator? In the simprecicchian Config.plist the only entry having to do with UIScale is under the "Delete" section under NVRAM.

simprecicchiani, 2022-02-08:
I think UIScale got its own configuration key in a recent OpenCore release. Before then was defined under NVRAM. You can follow the documentation and change it accordingly.

cyberfunk, 2022-02-08:
Thank you for your prompt response and for providing this excellent EFI! I manage to track down an example config.plist and implement the changes accordingly. You have to put this entry:

   Key                   Class             Value
UIScale             Number               1

Under UEFI>Output> and that fixes the issue. 

Also you can change "AdviseWindows" to "AdviseFeatures" in the Config.plist if you'd like to get rid of that OCS message on the Lenovo Boot screen that says: 

"OCS: No schema for AdviseWindows at 0 index, context >Generic>!"

This is for OpenCore 0.7.6