simprecicchiani/ThinkPad-T460s-macOS-OpenCore #150: Bluetooth Fix

meshackjr, 2022-04-04:
Hello, i am having trouble with bluetooth on Monterey, i have read from open core about the [rewrite of bluetooth stack from apple](https://dortania.github.io/OpenCore-Install-Guide/extras/monterey.html#bluetooth) but i cant seem to apply the patches on my own. Currently i cant connect to bluetooth devices from my laptop. Help please

559911, 2022-04-25:
[image]
try this https://openintelwireless.github.io/IntelBluetoothFirmware/FAQ.html#what-does-this-kext-do

mortuzahossain, 2022-11-08:
I use version [0.7.6 of EFI](https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore/tree/0.7.6) config everything works except Bluetooth. Although in the mentioned whats working section it is listed as working. I installed 12.6.1 macOS. Is there any fix available.

@559911 In the version 0.7.6 EFI config there is already present `IntelBluetoothFirmware.kext` 

Do i need to change it? 

Thanks in advance.

mintyleaf, 2022-11-27:
seems like this is __the fix__
@simprecicchiani

for about hour no issues with bluetooth since reboot with that option enabled
`UEFI -> ProtocolOverrides -> HashServices -> true`
https://www.tonymacx86.com/threads/volume-hash-mismatch-error-on-macos-monetery.319806/post-2316919

null2264, 2023-01-28:
I found a fix for L460, it has similar specs as T460(s), so maybe it'll help?

https://github.com/null2264/L460-Hackintosh#bluetooth-workaround

I basically set the `BT_DEVICE_ADDRESS` environment variable to my Bluetooth Controller MAC Address as soon as macOS boot (via `/etc/zshenv`), and somehow that fix it