simprecicchiani/ThinkPad-T460s-macOS-OpenCore #152: Bluetooth turns on but keeps loading

thomas-catt, 2022-05-21:
**macOS Monetery, T460 with Intel Bluetooth & Wi-Fi card.** When I turn on bluetooth, a loader appears and keeps spinning in the control center.

[image]

If I open bluetooth preferences, the window shows that bluetooth is turned off, and the *Turn Bluetooth On* button doesn't work at all.

[image]

Bluetooth is completely non-functional. I have an Intel Bluetooth & Wi-Fi card, and am using the Intel WLAN fixes (Wi-Fi works fine) as well. But I noticed that somewhere in *System Information*, my Bluetooth card is being listed as a ***Broadcom*** one (showing "BCM"). Could this be related?

[image]

 I tried to get rid of Broadcom by editing my plist, but this won't go anywhere. There are more Broadcom traces if I search with the *IORegistryExplorer* app: 

[image]

Any help will be appreciated. Btw thanks for the EFI, mostly this EFI has minimal bugs, I was successful in fixing a few of them, but unfortunately can't find a fix for this.

marcfusch, 2022-06-06:
Apple rewrote the bluetooth stack in monterey, breaking compatibility with many devices (mainly old audio codecs).
I went back to Big Sur flawlessly with time machine and bluetooth finally worked again.

victorovento, 2022-06-11:
I have the same problem

thomas-catt, 2022-06-12:
I found someone on reddit who had fixed this problem on their machine willing to help. I'll try to apply thise fixes and update

thomas-catt, 2022-06-12:
accidentally closed the issue sorry

thanosz, 2022-06-12:
> I found someone on reddit who had fixed this problem on their machine willing to help. I'll try to apply thise fixes and update you guys

Thank you!

cyberfunk, 2022-06-16:
If you have issues going forward I highly recommend the Broadcom BCM94360CS2 Wi-Fi and Bluetooth Card and the required adapter. I have a Lenovo Yoga 260 and this baby works as well or better than a native MacBook Wifi+BlueTooth setup. Location services Find My Mac etc. I tried Monterey and it worked flawlessly there as well but have returned to Big Sur due to issues with making bootable clones in Monterey.

victorovento, 2022-08-16:
this worked for me 
https://youtu.be/b4ettyKkcFg

thomas-catt, 2022-09-24:
Fixed the issue, see the edit for solution.

Closing.

closed 2022-09-24 by thomas-catt

thomas-catt, 2026-03-06:
> [@thomas-catt](https://github.com/thomas-catt) I read your edit but i think you mistyped. The fix is to load lilu before or after BluetoolFixup?

I haven’t dealt with this issue in a long time now but if I remember correctly you’re right I made a typo. Lilu must load before any of its plug-ins. 

I think if you’re on ProperTree, a clean snapshot (Ctrl/Cmd + Shift + R) fixes the kext loading order. If not, give ocvalidate a try.