simprecicchiani/ThinkPad-T460s-macOS-OpenCore #159: Upgraded to venturae and now its stuck on apple logo with half Installed

YadaDevv, 2022-11-04:
what should i do?

duszmox, 2022-11-04:
It was a bold move. My best guess would be to get a usb stick and another computer. Download the latest EFI from this repo, then update it. You can find the official guide on the dortania.github.com site. Then you should also update your Kexts. There is a link for every one of the kexts in the README.md. You should just replace them with the newly downloaded ones. And then you should boot the laptop from the USB. And voalà.

YadaDevv, 2022-11-04:
actually it was not my move, i was at work and sis wanted to use my laptop by the time i was home she explained what she did....
if you help me i really appreciate it! i dont know how to upgrade kexts and those sutff everytime a new update comes i come here to this repo and do what u guys say! all of my stuff are there and there is some codes i've written last night that is not push'ed to GitHub

duszmox, 2022-11-05:
I updated the EFI, since I don't have a T460s anymore I'm unable to test it, but hopefully it should work. I created a pull request, but also you can download it from the [fork of my repo](https://github.com/duszmox/Thinkpad-T460s-macOS-OpenCore/releases/tag/0.8.5). Hope it works

YadaDevv, 2022-11-05:
sadly its going into a bootloop

simprecicchiani, 2022-11-05:
> I updated the EFI, since I don't have a T460s anymore I'm unable to test it, but hopefully it should work. I created a pull request, but also you can download it from the [fork of my repo](https://github.com/duszmox/Thinkpad-T460s-macOS-OpenCore/releases/tag/0.8.5). Hope it works

As always, many thanks for your contribution!

@YadaDevv so sorry to hear the sad event. @duszmox's update [is out](https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore/releases/tag/0.8.5). I hope you'll be able to fix your issue.

YadaDevv, 2022-11-07:
okay so thanks for the help! i finished the upgrade by using the new EFI u guys just said and i changed a bit here and there finally i am on MacOS 13, but the about section is showing wrong infos for example i have an i7 it says i have i5! and in during the boot i get these "No schema LegacyEnable at 2 index, context <NVRAM>! "  and the system is so laggy! 
[image]

duszmox, 2022-11-07:
Oh boi, it seems graphic acceleration is broken. I have to do some research

YadaDevv, 2022-11-08:
Okayyyy so just help me on this problem and i will publish the whole EFI folder for you guys after that! i dont know how to set the PCI root's Intel Graphic stuff! when i set the correct APPLE, ig-platform-id it wont boot! if i write a nonsense one like 012345 it will boot and says i have 7MB VRAM, if i set none i get a black screen, if i set the correct one the bar gets to 1/2 and freezes!