simprecicchiani/ThinkPad-T460s-macOS-OpenCore #163: New EFI leads to missing Boot option in OpenCore Picker.

Kerberos124, 2022-11-08:
Hi,
I successfully installed MacOS on my Thinkpad in the past and it is running perfectly on MacOS BigSur (11.1). I would like to update my System to a newer MacOS Version, so I ported over my SMBIOS data from the old EFI Config to the new one. But after trying it on an USB stick only the EFI hdd appeared as a Boot option in the OpenCore Picker. This Option clearly does not work as there is nothing else than the OpenCore Boot-loader on the Usb Stick.
What can I do so OpenCore can Find my MacOS SSD like it does on the old EFI version?
Thanks for any tipps in advance!

aptonline, 2023-01-02:
I’m seeing the same issue with this EFI.  Did you manage to resolve the issue?

Kerberos124, 2023-01-04:
No, I did not. I am still on the old EFI Version so I can use the System.

aptonline, 2023-01-06:
> No, I did not. I am still on the old EFI Version so I can use the System.

Out of interest which EFI version are you using?