simprecicchiani/ThinkPad-T460s-macOS-OpenCore #172: Wi-Fi Not Working on Ventura 13.0

romyee, 2022-12-30:
Just recently installed Ventura 13.0 on my Lenovo T460s rig under
[EFI-T460s-OP.zip](https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore/files/10323790/EFI-T460s-OP.zip)
 Opencore 0.87 with the following specs:

i5-6300U Processor
14.0" FHD
8 GB DDR4 2133MHz
Intel HD Graphics 520 integrated
M.2 SSD, SATA Intel 256GB OPAL 2.0
Intel Dual Band Wireless-AC 8260, 2x2, Wi-Fi + Bluetooth® 4.1

Everything seems to be working except for the Wi-Fi. Used the guidelines & resources of this link, [ThinkPad-T460s-macOS-OpenCore](https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore). Used the latest Airportitilwm.kext. Wi-Fi icon shows up but no network is shown when scanning. My EFI is attached. Some kext files are dropped to cut down on the file size. 
[EFI-T460s-OP.zip](https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore/files/10323798/EFI-T460s-OP.zip)
Can anyone help?

null2264, 2023-01-01:
If what you mean by WiFi not working is WiFi not scanning any network then it's a known itlwm issue https://github.com/OpenIntelWireless/itlwm/issues/799

You might be able to temporarily fix it by running `sudo pkill airportd` (https://github.com/OpenIntelWireless/itlwm/issues/799#issuecomment-1368108175)

PeiFengBin, 2023-02-26:
> If what you mean by WiFi not working is WiFi not scanning any network then it's a known itlwm issue [OpenIntelWireless/itlwm#799](https://github.com/OpenIntelWireless/itlwm/issues/799)
> 
> You might be able to temporarily fix it by running `sudo pkill airportd` ([OpenIntelWireless/itlwm#799 (comment)](https://github.com/OpenIntelWireless/itlwm/issues/799#issuecomment-1368108175))

I have the same problem.But I `sudo pkill airportd` doesn't work.
Used this releases 0.8.5 EFI.
i5-6300U Processor
Intel Dual Band Wireless-AC 8260, 2x2, Wi-Fi + Bluetooth® 4.1

null2264, 2023-02-26:
It sometimes took a few attempts and/or require you to wait a few minutes for some reason... Honestly, if you could I suggest downgrading to (or staying on) Monterey for now, there are just too many issues on Ventura. Unless there's a feature from Ventura you actually rely on.

PeiFengBin, 2023-02-26:
> It sometimes took a few attempts and/or require you to wait a few minutes for some reason... Honestly, if you could I suggest downgrading to (or staying on) Monterey for now, there are just too many issues on Ventura. Unless there's a feature from Ventura you actually rely on.

Thank you for your answer. I'm considering downgrading to Monterey.