simprecicchiani/ThinkPad-T460s-macOS-OpenCore #180: Maintaners please update kext for itlwm (for Ventura)

qiloz, 2023-07-08:
@simprecicchiani 
## Based on #172 issue
Finally Wi-Fi is working properly on v2.2.0 of itlwm on Ventura
https://github.com/OpenIntelWireless/itlwm/releases/tag/v2.2.0 (AirportItlwm_v2.2.0_stable_Ventura.kext.zip) 
The issue with Wi-Fi is fixed here: https://github.com/OpenIntelWireless/itlwm/issues/799
---
### Please update file AirportItlwm-13.kext with replacing to new patch of itlwm to same name 🙏

[image]

TheMemeSniper, 2023-07-21:
to other people who have this same issue: drop the latest release of the kext over your already existing (old and broken) one. it works perfectly fine without issue for me

simprecicchiani, 2023-07-26:
Everyone is welcome to fork the repo and create a pull request