simprecicchiani/ThinkPad-T460s-macOS-OpenCore #188: Unable to make work in Sonoma, boot looping.

whalelinguni, 2024-04-29:
The only difference with my t460s is I have a 1440 LCD and no touch. I can get into an past OC fine, but when the apple logo appears and the loading bar gets maybe 15% done, system reboots for Sonoma. 

From Hardware Compatibility in the github Readme:
2. 1440p displays should change NVRAM -> Add -> 7C436110-AB2A-4BBB-A880-FE41995C9F82 -> UIScale:2 to get proper scaling while booting.

However, I'm uncertain if this is correct? Opening the plist in OCAT I get a warning for another error. But I do not see UIScale where it is described. I tried adding, did not make a difference. Pic of OCAT: https://i.imgur.com/onrFfkL.png

OCAT also throws this error after loading the plist "UEFI->Output->InitialMode is illegal (Can only be Auto, Text, or Graphics)!
CheckUefi returns 1 error!" Which I have also tried setting to 'Auto' with and without UIScale added and set to '2' under NVRAM 7C43.....C9F82 did not change anything either way, still rebooting when the loading bar on the apple logo gets maybe 15% or so. 

Attempted to use Ventura, which boots into MacOS recovery without any issues. Since the wifi isn't working, I tried to apply the Ventura installer dmg image to my nvme, which applies without any issues. However I cannot boot at all. Acting like there's no EFI boot on the drive, doesn't attempt anything. 

I'm new to this, so I don't really know fully what I am doing. Just thought I would post this here if anything to relay some issues I have had and maybe someone down the line may find this information useful for reasons. 

Running OC 0.9.9 tried with Sonoma and Ventura. Laptop is a t460s same specs as the readme, but non-touch and 1440p LCD. Have verified all bios settings match the settings described in the readme.

Anarooo, 2024-07-27:
Not sure if you've abandoned hope on this issue. I was having this exact issue where it looped on Monterey and I've managed to fix it. I just posted the fix here: https://github.com/simprecicchiani/ThinkPad-T460s-macOS-OpenCore/issues/189

It's possible it could work for Sonoma, I haven't tested it yet as I have yet to upgrade.

whalelinguni, 2024-07-29:
> Not sure if you've abandoned hope on this issue. I was having this exact issue where it looped on Monterey and I've managed to fix it. I just posted the fix here: #189
> 
> It's possible it could work for Sonoma, I haven't tested it yet as I have yet to upgrade.

Oh awesome thank you. I will check that out. Its possible my ram was an issue as well at the time. Shortly after I put my win10 nvme back in then that started locking up like crazy. Bad ram,  I feel like it would bootloop on me as well some, I just recall thinking how could 2 different drives with 2 different os in any way be doing the same thing.

basharalbashier, 2024-08-27:
if you solved your issue ; please tell me how?